DrTon/jMAVSim #12: Opengraphscene Port?

catch-twenty-two, 2015-02-09:
Hi Anton, is there anyreason why you chose to use Java3d?  It seems like it is on it's way out, I was considering trying to convert your source code to Opengraphscene/C++.  Would you be availible for questions while I was attempting this undertaking? Thanks

DrTon, 2015-04-18:
Hi! Yes, of course. Moreover would be good to support 2D or even headless operation. Need to move all vectors/matrices to other library, I can help with it. Do you know any good library for vectors/matrices etc? I tried ejml and la4j. la4j looks better for me - better interface, but not sure about performance, need to test more.

jjjskycatch, 2015-04-18:
Hi Anton, I was  talking about porting everything to C++.  I'm sure the performance would be very good. I agree 2d and especially headless operation would be very useful.

This is the library I had in mind, it is supported by the open source community and many commericial products and runs across all platforms:

http://www.openscenegraph.org/

DrTon, 2015-05-18:
@jjjskycatch, I tried to lean more about scene graph and 3D libraries for java. Looks like java3d is not so bad, it uses also scene graph object model, just need to be careful with native libs bindings, as with any other 3D library. Porting everything to C++ looks interesting, but I have no time for this. If you will do it, I can help with physics, but not with visualization and GUI part. Java looks good for me, it saves a lot of programming time compared to C++.