DrTon/jMAVSim #13: Exception in thread "main" java.nio.BufferUnderflowException

1091286, 2015-03-03:
Hi.
I'm trying to use your simulator, but I'm getting this error when I run mavros (rosrun mavros px4.launch). On mavros I get one heartbeat, and then mavros looses connection with the simulator.
Can you help me?

Exception in thread "main" java.nio.BufferUnderflowException
	at me.drton.jmavlib.mavlink.MAVLinkMessage.<init>(MAVLinkMessage.java:82)
	at me.drton.jmavlib.mavlink.MAVLinkStream.read(MAVLinkStream.java:39)
	at me.drton.jmavsim.UDPMavLinkPort.update(UDPMavLinkPort.java:66)
	at me.drton.jmavsim.MAVLinkConnection.update(MAVLinkConnection.java:44)
	at me.drton.jmavsim.World.update(World.java:33)
	at me.drton.jmavsim.Simulator.run(Simulator.java:159)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:132)
	at me.drton.jmavsim.Simulator.main(Simulator.java:168)

FabioArnez, 2015-04-10:
same problem...

DrTon, 2015-05-15:
Fixed in https://github.com/DrTon/jMAVSim/pull/15. Test please!

@FabioArnez, it's messages from autopilot I think, so need to find who sends this command first.

DrTon, 2015-05-18:
Fixed in master, test please.

closed 2015-05-18 by DrTon