DrTon/jMAVSim #17: Error in Windows

Vavooon, 2015-05-25:
Every time i run jMAVSim this error appears:

Exception in thread "main" java.net.BindException: Cannot assign requested address: connect
	at sun.nio.ch.Net.connect0(Native Method)
	at sun.nio.ch.Net.connect(Net.java:465)
	at sun.nio.ch.DatagramChannelImpl.connect(DatagramChannelImpl.java:738)
	at me.drton.jmavsim.UDPMavLinkPort.open(UDPMavLinkPort.java:36)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:112)
	at me.drton.jmavsim.Simulator.main(Simulator.java:145)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:606)
	at com.intellij.rt.execution.application.AppMain.main(AppMain.java:140)

Process finished with exit code 0

DrTon, 2015-05-27:
It's something bad with UDP port on windows. If you don't need to connect ground station you can comment this line https://github.com/DrTon/jMAVSim/blob/master/src/me/drton/jmavsim/Simulator.java#L112

I will investigate the issue and let you know, thanks for the reporting!

Vavooon, 2015-05-27:
I want to control UAV from CGS so i need to connect :)
Ok, thanks for the help.

Shaun413, 2015-05-27:
I have the same exact Issue as well. Please let me know if you find a way to fix this.

DrTon, 2015-05-27:
Fixed in https://github.com/DrTon/jMAVSim/commit/6ec80bf643bb1230790cdb153754e5adc24feae8, try please and close issue if it works for you.

Vavooon, 2015-05-28:
It works, thanks.

closed 2015-05-28 by Vavooon