DrTon/jMAVSim #18: GPS signal found/lost and OFFBOARD reject

weiweikong, 2015-05-25:
## GPS signal found/lost
Is there anyreason why I got a warning 
```
MSG: [inav] GPS timeout
```
or
```
MSG: [inav] init ref: 55.7533950, 37.6254270,   0.0000
MSG: home: 55.7533950, 37.6254270, -0.01
```
then 
```
(MAV 001:50) [inav] GPS signal found 
(MAV 001:50) [inav] GPS signal lost 
```
these two warnings when I started jMAVsim and keep warning all the time even though I armed the plane with RC.

My setting up are
- Ubuntu 14.04
- Pixhawk with Pixhawk Fireware
- QGC < - udp -> jMAVSim < - usb(/dev/ttyACM0) -> Pixhawk

## OFFBOARD reject
I tried both ```RC_MAP_OFFB_SW = 8 ``` and ```rosrun mavros mavsys mode -c OFFBOARD``` method, but still got 
```
(MAV 001:50) REJECT OFFBOARD 
```
I checked the ros topic ```/mavros/setpoint/local_position``` worked fine but only send (0, 0, 3) for climbing.

My setting up are
- Ubuntu 14.04
- Pixhawk with Pixhawk Fireware
- QGC < - udp -> MAVROS<- udp -> jMAVSim < - usb(/dev/ttyACM0) -> Pixhawk
- ```./setpoint_demo.py``` as mentioned at https://pixhawk.org/dev/ros/hil.

DrTon, 2015-05-27:
```
(MAV 001:50) [inav] GPS signal found 
(MAV 001:50) [inav] GPS signal lost 
```
This may indicate that you have hardware GPS connected and disabling it in HIL mode doesn't work. It's known PX4 issue, report them please. As workaround just disconnect GPS module while simulating.

For rejecting OFFBOARD need to digg, PX4 may not receive offboard control topic correctly, or something else. Start from here https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L715 or search for "offboard" in commander.

Vavooon, 2015-05-28:
I have the same issue with GPS lost/found without connected GPS.

LorenzMeier, 2015-05-29:
@DrTon the publication disabling is not a known issue, but was quite recently introduced during a rework of the GPS driver (a week ago or so) on the release branch. Master 

@Vavooon: Use the stable branch which works fine now. Also make sure to read the offboard control development guide. https://pixhawk.org/dev/offboard_control

LorenzMeier, 2015-05-29:
@DrTon I propose to close this, as its not a jMAVSim issue.

LorenzMeier, 2015-05-29:
I filed this issue instead - note that stable / release_v1.0.0, the branches we run for Pixhawk, are not affected.
https://github.com/PX4/Firmware/issues/2249

Vavooon, 2015-05-29:
@LorenzMeier Sorry, but it seems you confused me with @weiweikong .
I don't use Offboard mode and use stable firmware.

Vavooon, 2015-06-03:
I tried previous version https://github.com/DrTon/jMAVSim/commit/b98862c6a5cf859e2dfba8b2691c339c94c2e139 and it works fine.

Vavooon, 2015-06-04:
@LorenzMeier I'm sorry, but my autopilot wasn't really updated to RC10 for some reason. 
It works fine.

alexanderceron, 2015-06-05:
Great Job in this HIL simulator. I need some help with offboard in jMAVSim. I don't know if somebody can thanks in advance.
I would like to connect jMAVSim with ros in offboard mode by using a launch file. The topics are listed but empty and the offboad mode is rejected. I am using 14.04 and indigo.  
After roslaunch running,  I can't see the Heartbeat message.

 I am able to connect jMAVSim with qgcontrol (last update from yesterday) and pixhawk and RC, by following the usage directions except the 7 step Check “Show Advanced Protocol Options” and enable “Emit Heartbeat” because I did not find that option.  I had to disconnect the GPS and callibrate sensors again.

I don't know if my problem is the udp configuration in launch file, I am trying to use my local host like this:
	<arg name="fcu_url" default="udp://:14554@localhost:14555" />
	<arg name="gcs_url" default="udp://:14556@localhost:14550" />

my /etc/hosts is
127.0.0.1	localhost
127.0.1.1	alex
192.168.1.1     friend

# The following lines are desirable for IPv6 capable hosts
::1     ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

weiweikong, 2015-06-07:
Hi, sorry for the late reply, recently, I have already tested with the 
- PX4 Firmware (upgraded by firmware update in QGC with `Standard Version (stable)`) 
- jMAVsim (72d353c04e). 

The GPS problem has been fixed. Still, I could not switch to `OFFBOARD` control.
 
## ☑ 1.  Pixhawk + jMAVSim + QGC+ ROS => HIL Waypoints
### 1.1 Baisc Setup
**Pixhawk** <- [microUSB( `/dev/ttyACM0`)] -> **jMAVSim** <- [udp(`14555`)] -> **ROS Nodes**<- [udp(`14556`)] -> **QGC**
#### 1.1.1. Configuration of jUAVSim `Simulator.java`
```
serialMAVLinkPort.open("/dev/ttyACM0", 57600, 8, 1, 0);
udpMavLinkPort.open(new InetSocketAddress("127.0.0.1", 14555), new InetSocketAddress("127.0.0.1", 14550));
```
#### 1.1.2. Configuration of ROS Launch
- px4_pluginlists.yaml
```
     plugin_blacklist:
     - 'image_pub'
     - '3dr_radio'
     #- 'setpoint_position'
     - 'setpoint_attitude' 
     - 'setpoint_accel'
     - 'setpoint_velocity'
     - 'actuator_control'
```
- px4.launch
```
    <arg name="fcu_url" default="udp://:14550@localhost:14555" />
    <arg name="gcs_url" default="udp://:14557@localhost:14556" />
```

> **Question 1**: here is a little different with the Pixhawk-HIL Wiki (https://pixhawk.org/dev/ros/hil). With my private setting, I could check the data, for example, `mavros/imu/data` by using `rostopic echo /mavros/imu/data` and there is `CON: Got HEARTBEAT, connected`. However, if I set the udp ports as the Wiki, there was a ROS Node, but no data flow. Maybe I confused with the udp setting, such as `bind_host` and `remote_host`. Some help?
 
#### 1.1.3. Configuration of QGC
    - `File` -> `Manage Communication Links` -> `Comm Links` 
    - `Add` -> `UDP`  -> `Listenging Port` = 14556
    - `Add` -> `Target Hosts` -> `locoalhost:14557`
    - `Connect`

#### 1.1.4. Configuration of Pixhawk
    - `SYS_COMPANION` = 57600
    - `SYS_AUTOSTART` = 1001
> **Question 2**: In my opinion, the `SYS_COMPANION` should be 57600 due to the configuration, `serialMAVLinkPort.open("/dev/ttyACM0", 57600, 8, 1, 0);` , in `Simulator.java` file. Is that right ?

### 1.2 Waypoints Setting Steps
- Step 1: Run jUAVSim
```
java -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator
```
- Step 2: ROS Launch
```
roslaunch mavros px4.launch
```
- Step 3: Setting QGC as 1.1.3 mentioned.
- Step 4: ARM the Quadrotor
- Step 5: Set RC to `AUTO|MISSION`
- Step 6: Set the Waypoints using QGC
[image]

### 1.3 Comments
Above configurations work fine and stable. So I tried to test `OFFBOARD` mode.

## □ 2. Pixhawk + jMAVSim + QGC+ ROS => HIL `OFFBOARD`
### 2.1. The basic setup is the same as Section 1.
### 2.2. Run jUAVSim and ROS launch got
```
CON: Got HEARTBEAT, connected.
```
### 2.3. Confirmed the `SYS_COMPANION` = 57600 due to the `/dev/ttyACM0` device and `SYS_AUTOSTART` = 1001. 
[... 69 lines cut ...]
5 CURRENT POSITION XYZ = [  0.0000 ,  0.0000 , -0.1781 ]
6 CURRENT POSITION XYZ = [  0.0000 ,  0.0000 , -0.1551 ]
7 CURRENT POSITION XYZ = [  0.0000 ,  0.0000 , -0.1496 ]
 
DISABLE OFFBOARD MODE
 
READ SOME MESSAGES
Got message LOCAL_POSITION_NED (spec: https://pixhawk.ethz.ch/mavlink/#LOCAL_POSITION_NED)
    pos  (NED):  0.000000 0.000000 -0.156523 (m)
Got message HIGHRES_IMU (spec: https://pixhawk.ethz.ch/mavlink/#HIGHRES_IMU)
    ap time:     67796216
    acc  (NED):  -1.475761  0.721853 -9.688719 (m/s^2)
    gyro (NED):  -0.004324 -0.002157  0.002136 (rad/s)
    mag  (NED):   0.250519  0.283799  0.188447 (Ga)
    baro:        998.889954 (mBar)
    altitude:    120.226318 (m)
    temperature: 49.309998 C
 
CLOSE THREADS
 
CLOSE PORT
```
> Question 5: Whether above results means that the Pixhawk could working under `OFFBOARD` mode? And what is the different between this independent codes and the `mavros` based codes?

Any more help? @DrTon  @Vavooon  @LorenzMeier And I will read the codes of PX4 firmware and mavros for more detail?

alexanderceron, 2015-06-09:
Now it is not necessary to disconnect the GPS in order to use HIL?
As I see when I change the value to 1001 from 6001 or disconnect the GPS I have to recalibrate sensors again in order to fly. I don't know if this have any solution?
I am using a firmware from last tuesday by downloading a new version of QGC and as I see it is better than previous that I had in position control.  I would ask if the most recent firmware came with the QGroundcontrol and I have to download QGC again or that firmware is downloaded automatically from internet by QGC using the option update firmawe?
Thanks

weiweikong, 2015-06-09:
@alexanderceron In my setup, I did not connect GPS device with Pixhawk. Usually, I upgrade the firmware through QGC and choose the `stable` one.

DrTon, 2015-06-25:
>QGroundcontrol and I have to download QGC again or that firmware is downloaded automatically from internet by QGC using the option update firmawe?

qgc downloads firmware from internet each time, so even old qgc will flash fresh firmware.

DrTon, 2015-06-25:
>As I see when I change the value to 1001 from 6001 or disconnect the GPS I have to recalibrate sensors again in order to fly. I don't know if this have any solution?

Actually you don't need to switch to 1001 (HIL quad), HIL should work fine even with normal setups, e.g. 4001 for generic quad. Just make sure that LiPo is not connected, or no props mounted, just for safety :) HIL itself doesn't require any sensors calibration, because simulated sensors used anyway, but it may be a bug in PX4 so it does preflight check with real sensors even in HIL, not sure about this. @LorenzMeier, could you comment current state?

LorenzMeier, 2015-06-25:
It doesn't do preflight checks. And please make sure to use stable, not master. Master works in most cases, but in contrast to stable I'm not guaranteeing a smooth experience. All the QA efforts go into stable.

killerlayup, 2015-11-04:
@weiweikong 
have you solve your issue?

iceicehyhy, 2015-11-11:
HI @weiweikong,
Seems that QGC---->Mavros communication is not stable now, as the Connection will be lost as soon as you connect to Mavros.

iceicehyhy, 2015-11-11:
Hi @weiweikong & @killerlayup ,
It seems that I could switch to OFFBOARD MODE, by using the commands:
rosrun mavros mavsafety arm followed by  rosrun mavros mavsys mode -c OFFBOARD.
Once I run the demo app, I could see the quadrotor moving as the log shows:

iceicehyhy@iceicehyhy:~/catkin_ws$ ./setpoint_position_demo
[INFO] [WallTime: 1447259162.811083] Climb
[INFO] [WallTime: 1447259196.818033] Sink
[INFO] [WallTime: 1447259203.223879] Fly to the right
[INFO] [WallTime: 1447259211.428319] Fly to the left
[INFO] [WallTime: 1447259219.632792] Fly in a circle
[INFO] [WallTime: 1447259276.546952] Fly home
[INFO] [WallTime: 1447259311.765495] Bye!

[ INFO] [1447259163.113779739]: FCU: Flight mode: OFFBOARD
[ERROR] [1447259312.281050825]: FCU: failsafe mode on
[ INFO] [1447259312.332545828]: FCU: Flight mode: LAND
[ INFO] [1447259312.380169976]: FCU: [mpc] reset pos sp: 0, 0
[ INFO] [1447259312.432549006]: FCU: [mpc] reset alt sp: 0

I only face one problem now:
The communication between QGC and Mavros is not stable.

killerlayup, 2015-11-12:
@iceicehyhy  nope 
I have the same issue with yours

iceicehyhy, 2015-11-17:
@weiweikong  
hi, i saw you listen to some rostopic during the simulation.
May i know which topic you're listening to? I've attached a pic for you reference.
[image]

weiweikong, 2015-11-17:
Sorry @iceicehyhy , I did not catch your point.
Usually, I listen `mavros/local_postion/local` to get the location of the drones; also some other topics are needed which depend on your application.

iceicehyhy, 2015-11-17:
@weiweikong ,
thx, i got it.