DrTon/jMAVSim #19: Simulator is inactive for a minute after start

Vavooon, 2015-05-28:
Every time i start simulator i should wait almost minute, only after that time it writes 
Init MAVLink
and other messages and i am able to flight.

Vavooon, 2015-05-28:
It happens only on one of my PCs.

DrTon, 2015-05-28:
"Init MAVLink" should happen after 1s after receiving the first heartbeat from autopilot. How autopilot connected? Via USB? Try to add `System.out.println(msg)` before `sendMessage` here: https://github.com/DrTon/jMAVSim/blob/master/src/me/drton/jmavsim/SerialMAVLinkPort.java#L124
to see all incoming messages on serial port.

Vavooon, 2015-05-28:
Yes, it is connected via USB. I use latest rc10 firmware.
It doesn't depend on PC and happens on both my machines. 
Proposed debug command writes message only at the same time when "init mavlink" message appears.

DrTon, 2015-06-03:
Is it Windows? Maybe something bad with USB buffer (the data staying too long before flushing), I think I read something about this. Can you try on Linux or on another version of Windows?

Vavooon, 2015-06-03:
Ok, i'll try it on Linux later. It happens to me with January version (https://github.com/DrTon/jMAVSim/commit/b98862c6a5cf859e2dfba8b2691c339c94c2e139).
I can avoid this problem if i connect from QGC at first. After that jMAVSim connects in most cases.

Vavooon, 2015-06-04:
I tried it on Linux and all works fine.