DrTon/jMAVSim #20: Simulator can't  displaye the attitude

iidioter, 2015-05-30:
I use the pixhawk, and connect via USB (/dev/ACM0), and then the jMAVLink can't dispaly the status, just on the ground,

and i try to connect the QGC via udp, the QGC display No Rules.

Thanks.

iidioter, 2015-05-30:
could you tell me how to config the jMAVLink/mavros/QGC?
thanks!

DrTon, 2015-06-03:
Step by step please :)
> I use the pixhawk, and connect via USB (/dev/ACM0)
Are you connecting jMAVSim to pixhawk via /dev/ACM0? If you are using Linux it's likely ttyACM0, not ACM0. Also make sure that you use PX4 (not default APM) firmware (flashed using qgroundcontrol).

> jMAVLink can't dispaly the status
What is jMAVLink? jMAVSim or MAVLink? Where you want to see the status? Simulator starts with vehicle on ground, to takeoff you need to press safety switch and arm.

> and i try to connect the QGC via udp, the QGC display No Rules.
Hm, never seen this message. Anyway need to fix connection to pixhawk first.

Do you see "Init Mavlink" message in console when jMAVSim starts?

iidioter, 2015-06-08:
Thanks, I forget to connect the RC.

closed 2015-06-08 by iidioter