DrTon/jMAVSim #24: modeling Rotor latency

kd0aij, 2015-06-29:
These are impulse response models constructed from log data collected using MatrixPilot firmware in an AeroQuad frame. 3S battery, 1000KV 2212 motors, 10x4.7 props, 680mm between motor centers.
 The Pitch moment of inertia is a little higher than the Roll moment.

Impulse response model: A exp(-t/tau) sin(omega t)
Delay values in plot headers are the estimated command->response latency for thrust and rotation rate, respectively.

[image]

This plot is for a HobbyKing microQuad with motor spacing of 220mm:
[image]

DrTon, 2015-07-08:
Very useful, thanks! But can you provide some more comments. What was input pulse? Roll or pitch control? Then why you are talking also about accelerations, and what means "roll accel"?
Also we see here how controller performs, i.e. response of controller+motor+prop to setpoint, not motor+prop itself. So ideally need to look at response to e.g. thrust=50% -> thrust=60% step. Or roll=0 -> roll=0.5.

kd0aij, 2015-07-08:
The impulse response models are constructed by finding the parameters A, tau and omega which best fit the logged input/output signals over an entire test sequence involving aggressive roll/pitch maneuvers. Inputs are individual motor throttle settings, outputs are angular acceleration, rate and tilt. 
I looked at angular acceleration because it should correlate directly to differential thrust.

This guy did some interesting tests of esc/motor/prop response: http://www.rcgroups.com/forums/showthread.php?t=1006721
He's showing about 100msec lag to peak response for a 100us PWM input step.

DrTon, 2015-07-10:
Ah ok, I thought about linear acceleration, not angle acceleration. Thanks for that article, also very useful! I will review the defaults.