DrTon/jMAVSim #25: no main manifest attribute, in out/production/jmavsim.jar

kd0aij, 2015-06-29:
latest master:
```
01:31:43 {master} ~/git/drton/jMAVSim$ ant clean
Buildfile: /home/markw/git/drton/jMAVSim/build.xml

clean:
   [delete] Deleting directory /home/markw/git/drton/jMAVSim/out/production

BUILD SUCCESSFUL
Total time: 1 second
04:58:52 {master} ~/git/drton/jMAVSim$ ant
Buildfile: /home/markw/git/drton/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /home/markw/git/drton/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 58 source files to /home/markw/git/drton/jMAVSim/out/production/jMAVSim

jmavsim:
      [jar] Building jar: /home/markw/git/drton/jMAVSim/out/production/jmavsim.jar

all:

BUILD SUCCESSFUL
Total time: 4 seconds
04:58:58 {master} ~/git/drton/jMAVSim$ java -jar out/production/jmavsim.jar
no main manifest attribute, in out/production/jmavsim.jar

05:02:56 {master} ~/git/drton/jMAVSim$ git log
commit e224a4fa4ac77cae2f0c793bfeea48c0de938dea
Author: Anton Babushkin <anton.babushkin@me.com>
Date:   Mon Jun 29 22:34:55 2015 +0200

    Rotor: time constant scale fixed, documentation added

```

DrTon, 2015-07-08:
Let me check, thanks for the reporting.

DrTon, 2015-07-08:
Ah, no, it's ok. RTFM:
>Run:
>java -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator

Don't use -jar to run jmavsim.jar. This is because JOGL library is platform-dependent and it's not possible to pack all libs in one jar because of conflicts. Need to add separate target for each platform or solve filenames conflicts in JOGL somehow. Both ways IMO are difficult and not needed, because jMAVSim is only used by developers, who can write even long command to one line sh script :)

closed 2015-07-08 by DrTon

kd0aij, 2015-07-08:
@DrTon Thanks, since the FM only addresses OSX, I've been building executable JARs for Linux the same as I've always done for my own Java projects. It worked till recently, and I didn't understand the issues with JOGL.