DrTon/jMAVSim #26: Error With Simulator.class

javeryk21, 2015-07-23:
I am able to download and compile the jMAVSim directory using ANT.  When I run "java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -udp 127.0.0.1:14560" I get the error could not find or load main class me.drton.jmavsim.Simulator.  Am I not doing something correctly with the build.xml file?  Is there something else that I am missing here?

Thanks,

DrTon, 2015-11-10:
- Have you build it with 'ant' command?
- Do you have file out/production/jmavsim.jar?
- Maybe something wrong with classpath (-cp option), on different platforms format is different

nanthony21, 2016-10-13:
Did this ever get figured out? I have the same issue. It seems like "-Djava.ext....." should have a space: "-D java.ext..." but then I just get "Error: Could not find or load main class java.ext.dirs="