DrTon/jMAVSim #28: can't communcate through udp port

alsadi007, 2015-08-31:
I am trying to communicate to vagrant through udp , but I get the below message: 

FAHADs-MacBook-Pro:jMAVSim fahadkhamis$ java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -udp 127.0.0.1:14560
3D [dev] 1.6.0-pre12-daily-experimental daily

Exception in thread "main" java.io.IOException: jssc.SerialPortException: Port name - /dev/tty.usbmodem1; Method name - openPort(); Exception type - Port not found.
	at me.drton.jmavsim.SerialMAVLinkPort.open(SerialMAVLinkPort.java:38)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:110)
	at me.drton.jmavsim.Simulator.main(Simulator.java:146)
Caused by: jssc.SerialPortException: Port name - /dev/tty.usbmodem1; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:165)
	at me.drton.jmavsim.SerialMAVLinkPort.open(SerialMAVLinkPort.java:35)
	... 2 more

bidai541, 2015-09-20:
hill man, I met this question too, how did you deal with this problem?

alsadi007, 2015-09-20:
Hi mate,
I tried different things but I couldn't make it run. The software needs to be modified.Just decided then to use hardware in loop simulation

bidai541, 2015-09-20:
Is it essential in HITL mode that all the sensors should be connected to px4?

alsadi007, 2015-09-20:
I have done the simulation once and I don't really remember if I had the sensors connected. I believe   All the sensors data will be simulated.

bidai541, 2015-09-20:
OK. Thanks very much!  ^-^

bidai541, 2015-09-21:
Hi, I have know the reason why sitl cannot connected by jmavsim. You should clone the jmavsim project from this web  https://github.com/PX4/jMAVSim.git (not https://github.com/DrTon/jMAVSim).  So the readme file is wrong.

alsadi007, 2015-09-21:
was you able to compile it ? 
I have tried this before but I got this error when I tried to compile it : 

compile:
    [javac] Compiling 58 source files to /Users/fahadkhamis/git_folder/jMAVSim/out/production/jMAVSim
    [javac] /Users/fahadkhamis/git_folder/jMAVSim/src/me/drton/jmavsim/UDPMavLinkPort.java:99: cannot find symbol
    [javac] symbol  : method getLoopbackAddress()
    [javac] location: class java.net.InetAddress
    [javac]             possibleAddress = InetAddress.getLoopbackAddress();
    [javac]                                          ^
    [javac] /Users/fahadkhamis/git_folder/jMAVSim/src/me/drton/jmavsim/UDPMavLinkPort.java:129: cannot find symbol
    [javac] symbol  : method getRemoteAddress()
    [javac] location: class java.nio.channels.DatagramChannel
    [javac]             SocketAddress remote = channel.getRemoteAddress();
    [javac]                                           ^
    [javac] 2 errors

BUILD FAILED
/Users/fahadkhamis/git_folder/jMAVSim/build.xml:20: Compile failed; see the compiler error output for details.

bidai541, 2015-09-21:
May be your jdk version is not right. I installed "jdk-8u60-linux-x64.tar.gz"  from here http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

docHell, 2019-03-05:
I've created a virtual machine running Xubuntu 18.04.
I've correctly ( I suppose ) installed jmavsim and px4.
When I digit "sudo make posix_sitl_default jmavsim" the simulator starts correctly and I can run some commands succesfully like "commander takeoff".

According to the instructions, to connect to the wifi I must run the simulator using a different command :

sudo make broadcast jmavsim

Then I've to set manually a parmeter in the consolle the firtst time.

param set MAV_BROADCAST 1

The screen log is this :

ninja: Entering directory `/home/dronetest/Firmware/build/posix_sitl_broadcast'
[0/1] cd /home/dronetest/Firmware/buil...st/Firmware/build/posix_sitl_broadcast
SITL ARGS
sitl_bin: /home/dronetest/Firmware/build/posix_sitl_broadcast/bin/px4
rcS_path: posix-configs/SITL/init/ekf2
debugger: none
program: jmavsim
model: none
src_path: /home/dronetest/Firmware
build_path: /home/dronetest/Firmware/build/posix_sitl_broadcast
empty model, setting iris as default
SITL COMMAND: /home/dronetest/Firmware/build/posix_sitl_broadcast/bin/px4 /home/dronetest/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/dronetest/Firmware/test_data
INFO [Unknown] Creating symlink /home/dronetest/Firmware/ROMFS/px4fmu_common -> /home/dronetest/Firmware/build/posix_sitl_broadcast/tmp/rootfs/etc

| ___ \ \ \ / / / |
| |/ / \ V / / /| |
| __/ / \ / /| |
| | / /^\ \ ___ |
_| / / |_/

px4 starting.

INFO [Unknown] Calling startup script: bash etc/init.d-posix/rcS 0
INFO [dataman] Unknown restart, data manager file './dataman' size is 11405132 bytes
INFO [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Buildfile: /home/dronetest/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

libEGL warning: DRI2: failed to authenticate
Init MAVLink
INFO [simulator] Got initial simulation data, running sim..
INFO [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
INFO [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
INFO [logger] logger started (mode=all)
INFO [logger] Start file log
INFO [logger] Opened log file: ./log/2019-03-05/01_06_01.ulg
INFO [Unknown] Startup script returned successfully
pxh> INFO [mavlink] using network interface enp0s3, IP: 192.168.1.200
INFO [mavlink] with netmask: 255.255.255.0
INFO [mavlink] and broadcast IP: 192.168.1.255
INFO [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO [ecl/EKF] EKF commencing GPS fusion
ERROR [Unknown] WARNING! TIME IS NEGATIVE! 158155213 vs 158155216

I've tried with some software ( in java or nodejs) to connect, inside and outside the vm, but the " ECONNREFUSED" is always the result.

I've tried also to ping using nmap the port, but

for the 14556/7

sudo nmap -p 14556 -sU 192.168.1.200

Starting Nmap 7.60 ( https://nmap.org ) at 2019-03-05 02:53 CET
Nmap scan report for 192.168.1.200
Host is up (0.00017s latency).

PORT STATE SERVICE
14556/udp open|filtered unknown
MAC Address: 08:00:27:66:94:F7 (Oracle VirtualBox virtual NIC)

Nmap done: 1 IP address (1 host up) scanned in 0.93 seconds

instead for the 14550/40:

sudo nmap -p 14540 -sU 192.168.1.200

Starting Nmap 7.60 ( https://nmap.org ) at 2019-03-05 02:53 CET
Nmap scan report for 192.168.1.200
Host is up (0.00012s latency).

PORT STATE SERVICE
14540/udp closed unknown
MAC Address: 08:00:27:66:94:F7 (Oracle VirtualBox virtual NIC)

Nmap done: 1 IP address (1 host up) scanned in 0.82 seconds

I'm trying now using "https://github.com/kvenux/nodegcs"

But the result is the same : ECONNREFUSED

Can you help me ?