DrTon/jMAVSim #30: jMAVSim does nothing

zshn25, 2015-11-02:
This is my first time using any of these. Please tell me if I did anything wrong.

I started jMAVSim using 
```
sudo java -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator
```
Then, I connected the quad. I run
```
roslaunch mavros px4.launch
```
Then I armed it by
```
rosrun mavros mavsafety arm
```
I pressed the safety switch and armed it. There was no change in the simulator window. The output of the jMAVSim Terminal is 
```
3D [dev] 1.6.0-pre12-daily-experimental daily

libEGL warning: DRI2: failed to authenticate
Exception in thread "main" java.io.IOException: jssc.SerialPortException: Port name - /dev/tty.usbmodem1; Method name - openPort(); Exception type - Port not found.
	at me.drton.jmavsim.SerialMAVLinkPort.open(SerialMAVLinkPort.java:38)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:110)
	at me.drton.jmavsim.Simulator.main(Simulator.java:146)
Caused by: jssc.SerialPortException: Port name - /dev/tty.usbmodem1; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:165)
	at me.drton.jmavsim.SerialMAVLinkPort.open(SerialMAVLinkPort.java:35)
	... 2 more
```
Please help!!

killerlayup, 2015-11-05:
hi
I have the same issue now!

zshn25, 2015-11-05:
Why is there this Warning? 
```
libEGL warning: DRI2: failed to authenticate
```

DrTon, 2015-11-10:
Sorry, I'm not using ROS, ask this in PX4 fork: https://github.com/PX4/jMAVSim

closed 2015-11-10 by DrTon