DrTon/jMAVSim #31: WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.

killerlayup, 2015-11-03:
Hi,
I have an issue when I start jMAVSim to communicate with px4 by running:

    java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -serial /dev/ttyACM0 921600 -qgc

it shows the blow log:

    WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.

is that related to MAV_SYS_ID?

DrTon, 2015-11-10:
I think yes.

closed 2015-11-10 by DrTon

1ee7, 2015-11-23:
Do you resolved？ I have meeted the same problem?

1ee7, 2015-11-23:
How did you resolved? I meet the same problem

Thomas10111, 2016-01-06:
In file MAVLinkHILSystem.java you have to add "" around sysId (2 times)

else if (!gotHeartBeat && sysId != msg.getInt("sysId")) {
                System.out.println("WARNING: Got heartbeat from system #" + Integer.toString(msg.getInt("sysId")) +
                    " but configured to only accept messages from system #" + Integer.toString(sysId) +
                    ". Please change the system ID parameter to match in order to use HITL/SITL.");

Then compile by calling "ant" again.