DrTon/jMAVSim #35: jMAVSim Multiple Vehicles

musclesinwood, 2016-03-14:
HI
Can someone please guide me how can I get two vehicles on qgroundcontrol using jMAVSim?
Currently jMAVSim is simulating a single vehicle on qgroundcontrol. WHat I want is simulating two vehicles using a single jMAVSim simulator. Can anyone please guide me how can I proceed with this issue?

regards
Omar

DrTon, 2016-04-08:
That should be possible. Just try to create another vehicle:

```
        // Create vehicle with sensors
        Vector3d gc = new Vector3d(0.0, 0.0, 0.0);  // gravity center
        AbstractMulticopter vehicle = new Quadcopter(world, "models/3dr_arducopter_quad_x.obj", "x", 0.33 / 2, 4.0,
                0.05, 0.005, gc);
        vehicle.setMass(0.8);
        // Sensors setup etc.
```

And create corresponding MAVLink object for it:

```
        MAVLinkHILSystem hilSystem = new MAVLinkHILSystem(schema, 2, 51, vehicle); // Use another SysID here so qgroundcontrol can understand that it's another system!!!
        connHIL.addNode(hilSystem);
```

And finally add to the "world":

```
        world.addObject(vehicle);
```

If you use HIL you also will need to create the second connection to the second autopilot, or another UDP port for SIL.

weiminshen99, 2018-03-31:
@DrTon,
Thanks for your answers. Could you please explain in which files should these new lines be added? 
Thanks!
Wingman

weiminshen99, 2018-03-31:
Hi, @DrTon ,
Sorry to bother you again. Here are the four lines I inserted in the class Simulator in the file Simulator.java:
       vehicle2 = buildMulticopter();
       hilSystem2 = new MAVLinkHILSystem(schema, 2, 51, vehicle2); 
       connHIL.addNode(hilSystem2);
       world.addObject(vehicle2);
The system compiles fine, but runs into some errors that I will send you in the next message.

weiminshen99, 2018-03-31:
Here is the error message partially printed in a small window at the right-bottom corner of the GUI:

WARNING: Got heartbeat from system#-1 but configured to only comm system#2. Please change the system ID parameter to match in STIL.
Init MAVLink
Init MAVLink

Please let me know how to proceed from here. Thanks!