DrTon/jMAVSim #42: running JMAVSim in vmware fails

itayguy, 2018-04-01:
81 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [dataman] Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11405132 bytes
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Buildfile: /home/itayguy/git/PX4Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x3e26d121, isOwner false, <7e758363, b619854>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.8
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:418)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:759)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:642)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:214)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:175)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
	... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f8eb00010b0, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f8e94020c70, refCount 1, unCloseable false]
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x4ce470aa, isOwner false, <4da7bb0f, 4e58036c>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.8
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:418)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:759)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:642)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:214)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:175)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
	... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f19b0006c40, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f1994020c70, refCount 1, unCloseable false]
^C
Exiting...
pxh> Shutting down
Restoring terminal
/home/itayguy/git/PX4Firmware/Tools/sitl_run.sh: line 167: kill: (5238) - No such process
ninja: build stopped: interrupted by user.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Interrupt

Cody-Chen, 2018-04-21:
I have a same issue as you. ubuntu 16.04 and build on java8.  But I have fix it.
There are 2 errors on startup: 
1.
```
com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice
```
2.
```
com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1).
```
For the first one error:
Jmavsim have include a java3d lib on ```./lib```.
I update the lib file form http://jogamp.org/deployment/jogamp-current/archive/jogamp-all-platforms.7z

Second error.The  jogamp lib colud not find the right version of OpenGL, I find a patch here:
https://link.jianshu.com/?t=https%3A%2F%2Fjogamp.org%2Fbugzilla%2Fshow_bug.cgi%3Fid%3D1357
Just update the ```GLContextImpl.class``` . It is in the ```./lib/jogl-all.jar```.

signsmile, 2018-05-17:
the solution can not works.

signsmile, 2018-05-17:
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.8
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:440)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:214)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:175)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7f808c13ca10, refCount 1, unCloseable false]

signsmile, 2018-05-17:
``Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.8
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:440)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:214)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:175)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7f808c13ca10, refCount 1, unCloseable false]

signsmile, 2018-05-17:
`Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 17.2.8
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:440)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:765)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:648)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:586)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:214)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:175)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7f808c13ca10, refCount 1, unCloseable false]
`

profxerxes, 2018-05-18:
@signsmile  on VirtualBox, i enabled 3D acceleration; and voila the error disappeared.

ngupta9, 2018-07-16:
Had the same problem and figured that it had something to do with my graphics card due to similar errors upon looking online. Installed the latest nvidia drivers and started working! Had to make sure that in Software & Updates->Additional Drivers the correct drive was selected. Good luck!

pranavn248, 2018-08-09:
@Cody-Chen 
> com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice

Did you just add the "lib" folder from [here](http://jogamp.org/deployment/jogamp-current/archive/jogamp-all-platforms.7z), (i.e. /jogamp-all-platforms/lib) to the lib folder present in jMAVSim (/Firmware/Tools/jMAVSim/lib)?

docHell, 2018-08-13:
I had the same problem with a Oracle VM  xubuntu 18.04.
I've solved simply installing a new version of Open GL.

 `sudo add-apt-repository ppa:ubuntu-x-swat/updates`

and then 

 `sudo apt-get dist-upgrade`

I still have the problem :

libEGL warning: DRI2: failed to authenticate

but jMAVsim works.

MaEtUgR, 2021-01-18:
Thanks to @nwachukwucobinna for documenting his finding that 3D acceleration has to be enabled by the host!
I summarized the details for future reference in https://discuss.px4.io/t/jmavsim-fails-to-run-under-vmware-ubuntu/6242/5

I think we can close this issue.